geez0x1/CompliantJointToolbox #3: Handling of direct feedthrough in state space models

joernmalzahn, 2016-06-08:
The direct feedthrough is sometimes hardcoded to zero. That might cause problems. Sometimes the direct feedthrough is completely ommitted in the output vector. Commonly the direct feedthrough is denoted by D. However in getDynamicsMatrices the output variable D has a different meaning. That is confusing.

geez0x1, 2016-07-26:
Hardcoded zero fixed in 819283c671ca3188eeb4b0aed307d2d324790039 (didn't cause problems because in continuous-time, D is always zero)

geez0x1, 2016-07-26:
I agree that the definition of `D` in `getDynamicsMatrices` is confusing. An output like `A,B,C,D,I,D(damp),K` would be better. Any thoughts on how to rename the damping `D`?

joernmalzahn, 2016-11-13:
What about an 'R'? The matrix describes a resistive effect in the electroanalogous sense.

geezIIT, 2016-11-14:
That's nice. Let's do that.

joernmalzahn, 2016-11-14:
Would we then also renam all damping parameters? Otherwise we end up with a matrix R that contains a plenty of d_xy's.

geez0x1, 2016-11-14:
That's a good point.. I think renaming all the damping coefficients is less feasible. What if we simply rename it `D_damp`?

joernmalzahn, 2016-12-12:
Direct Feedthrough is implemented now. It is called D in the function signature. Damping matrix is R. Damping parameters are still d_xy's.

closed 2016-12-12 by joernmalzahn