geez0x1/CompliantJointToolbox #4: Review Templates

joernmalzahn, 2016-06-15:
Are all template files really necessary? Or are some obsolete?
* getStateSpace.m can be implemented in genericJoint class
* getNonlinearDynamics_multiple.m and getNonlinearDynamics.m seem to be redundant

geez0x1, 2016-06-15:
You're right on `getStateSpace()`. The other ones were introduced to ease the development of the jointBuilder a bit; it should be possible to merge `getNonlinearDynamics` and `getNonlinearDynamics_multiple`. I've assigned this bug to me.

geezIIT, 2016-06-16:
`getStateSpace()` has been removed and is implemented in `genericJoint`. As for the nonlinear dynamics implementations, all three files:

- `getNonlinearDynamics`
- `getNonlinearDynamics_multiple`
- `getNonlinearDynamics_none`

have been merged into a single `getNonlinearDynamics` template. Thus marking this as solved.

closed 2016-06-16 by geezIIT