geez0x1/CompliantJointToolbox #6: Position-dependent viscous friction model is unfinished

geezIIT, 2016-06-16:
Like the title states, the position-dependent viscous friction model is unfinished:
`model/nonlinear/viscous_pos_dep.m`

geez0x1, 2016-11-09:
Deleted from master in 29b83595b35479225094e40afe27349e01d220f4, now has its own branch, to be developed in due time.

joernmalzahn, 2016-11-13:
Hmm, this might go into the ripple model?

geezIIT, 2016-11-14:
Yeah it may be possible to merge this with the more unified ripple model as a variant that depends on speed, I'll have to see.

joernmalzahn, 2016-12-12:
Any update here?

geez0x1, 2016-12-28:
I thought about how to include it in the `torque_ripple` function but it was a bit unnatural, since it doesn't just 'oscillate' around zero mean (w.r.t. speed) but actually has an offset, unlike the torque ripple sources. I don't think we have an immediate need for this model though.