geez0x1/CompliantJointToolbox #15: Recovered Stateflow Charts

geez0x1, 2016-11-09:
The `Recovered Stateflow Charts` issue keeps popping up, despite the `deleteStateflowChart.m` code. 

Errors that can be seen while closing models relate to the same issue:
```
Warning: Error evaluating 'ModelCloseFcn' callback of SubSystem block (mask) 'my_model/JointModel'. 
 
Warning: Cannot open 'my_model' while it is being closed. 
 
Warning: Error evaluating 'ModelCloseFcn' callback of SubSystem block (mask) 'my_model/JointModel1'. 
 
Warning: Cannot open 'my_model' while it is being closed. 
```

joernmalzahn, 2016-11-13:
Does not look simple. Having a hard time to reproduce it. It might be a Simulink Version Issue. Will take some time, but I do agree, we have to fix this.

geez0x1, 2016-12-05:
@joernmalzahn Any progress on this?

joernmalzahn, 2016-12-05:
No.
    
    Am 05.12.2016 um 10:16 schrieb Wesley
      Roozing:
    
    
      @joernmalzahn
        Any progress on this?
      —
          it on GitHub, or mute
          the thread.

geez0x1, 2018-03-29:
This remains an issue for me even in R2017a.

joernmalzahn, 2018-07-27:
I have not seen any recovered stateflow char block in quite a while. @geez0x1 Is this still happening to you?

geez0x1, 2018-07-27:
Yes this still happens to me. A reliable way to reproduce it seems to be opening a model (that does not have orphaned StateFlow charts), simulating it, saving it, and closing it. Re-opening will then show an orphaned chart.

Removing the chart, saving, closing, and repeating a second time, appears to fairly reliably remove the orphaned chart.