geez0x1/CompliantJointToolbox #20: Position input as external disturbance for locked-output models

geezIIT, 2016-11-15:
Currently the external disturbance input is `tau_e`, the external torque acting on the link. In the cases where the link position is imposed (or, more narrowly, locked at zero), this input obviously does nothing.

An idea is to replace it with an imposed link position input for the `output_fixed` models. This is useful for testing things like actuator transparency. It can be achieved quite easily for the spring force by changing the state-space `B` matrix so that the second column contains something like `I\K`; however, this does not take into account the torsion bar internal damping. In fact, the torque on the motor from the torsion bar would become `-k_b * (q_m - q_l) - d_gl * q_m_dot`. So this is only approximate, although probably perfectly fine for most cases.

The only real solution is to have the external input be link velocity to obtain the damping part, and have a `q_l` state in the model to obtain the spring component.

@joernmalzahn any thoughts?

geezIIT, 2016-11-15:
(in the light of tagging v0.2 we might also move this to its own dev branch)

joernmalzahn, 2016-12-12:
Agree, it is useful and agree, maybe after tagging v0.2.

geez0x1, 2017-12-11:
This was added in daf43e4dd50074d42bae444fc49b59ad9e97b04f up to d26537644ae65ec34fdb3afadd2697b7116f3891, and part of v0.6.

closed 2017-12-11 by geez0x1