geez0x1/CompliantJointToolbox #21: Voltage limits

geez0x1, 2016-12-05:
Feature proposal; Adding voltage limits. Currently we only limit the torque, while based on the back-EMF and the max applicable voltage we could/should also limit it when it cannot be realised.

joernmalzahn, 2016-12-05:
I have the model for the electrical subsystem implemented. Just have
    to clean it and make it ready for the toolbox. This can do the job.
    
    Am 05.12.2016 um 10:18 schrieb Wesley
      Roozing:
    
    
      Feature proposal; Adding voltage limits. Currently we only
        limit the torque, while based on the back-EMF and the max
        applicable voltage we could/should also limit it when it cannot
        be realised.
      —
        thread.
          it on GitHub, or mute
          the thread.

geez0x1, 2016-12-05:
Does that include the electrical dynamics? If so, that significantly impacts the simulation speed. It's nice to have but we should also have the option to include voltage limits without simulating the full electrical dynamics.

joernmalzahn, 2016-12-05:
Yes, that is what I meant by "electrical subsystem". The part that 
affects the simulation speed a lot is the consideration of inductance 
that introduces small time constants of the system dynamics. In many 
cases these dynamics are fast enough to be neglected. In that case the 
electrical subsystem models the effect of back-emf, winding resistance 
and eventually also thermodynamics.

Am 05.12.2016 um 11:59 schrieb Wesley Roozing:
> dynamics? If so, that significantly impacts the simulation speed. It's 
> nice to have but we should also have the option to include voltage 
> limits without simulating the full electrical dynamics.

-- 
+++++++++++++++++++++++++++++++
Dr.-Ing. Jörn Malzahn

Istituto Italiano Di Tecnologia
Department of Advanced Robotics

Tel: +39 010 71781 598
Fax: +30 010 71781 232
Email: jorn.malzahn@iit.it

Via Morego 30
16163 Genova, Italy
+++++++++++++++++++++++++++++++

geez0x1, 2016-12-05:
OK, that's what I meant. In that case, feel free to add it to the toolbox (in its own branch please due to tagging v0.2 soon). I'm currently working on a unified torque ripple model which might make it into v0.2, depending on when #15 gets fixed.

joernmalzahn, 2016-12-12:
Electrical subsystem models are implemented now in genericJoint and as Simulink block. 
The genericJoint class has a new abstract method getElectricalDynamicsMatrices. Two implementations are available. One static model incorporating back-emf but neglecting winding inductance (used as default). One including the winding inductance (small simulation time steps needed).
The simulink block makes use of these functions and can simulate DAC quantization, communication delay and noise on top of that.

closed 2016-12-12 by joernmalzahn