geez0x1/CompliantJointToolbox #27: jointBuilderTest fails

joernmalzahn, 2017-03-07:
================================================================================
Error occurred in jointBuilderTest/testNoFrictionModels and it did not run to completion.

    ---------
    Error ID:
    ---------
    ''

    --------------
    Error Details:
    --------------
    Error using jointBuilder/buildJoint (line 99)
    jointBuilder.buildJoint error: Linear model 'full_dyn_no_friction' do not exist!
    
    Error in jointBuilderTest>testNoFrictionModels (line 127)
            jb.buildJoint(allParams{iPar},'full_dyn_no_friction');
================================================================================
.99                  error(['jointBuilder.buildJoint error: Linear model ''' modelName ''' do not exist!']);

================================================================================
Error occurred in jointBuilderTest/testCoulombModels and it did not run to completion.

    ---------
    Error ID:
    ---------
    ''

    --------------
    Error Details:
    --------------
    Error using jointBuilder/buildJoint (line 99)
    jointBuilder.buildJoint error: Linear model 'full_dyn' do not exist!
    
    Error in jointBuilderTest>testCoulombModels (line 148)
            jb.buildJoint(allParams{iPar}, 'full_dyn', 'coulomb');
================================================================================
.
================================================================================
Error occurred in jointBuilderTest/testMultipleNonlinearTerms and it did not run to completion.

    ---------
    Error ID:
    ---------
    ''

    --------------
    Error Details:
    --------------
    Error using jointBuilder/buildJoint (line 99)
    jointBuilder.buildJoint error: Linear model 'full_dyn' do not exist!
    
    Error in jointBuilderTest>testMultipleNonlinearTerms (line 173)
            jb.buildJoint(allParams{iPar}, 'full_dyn', {'coulomb', 'viscous_asym'});
================================================================================
.
================================================================================
Error occurred in jointBuilderTest/testInputParameters and it did not run to completion.

    ---------
    Error ID:
    ---------
    ''

    --------------
    Error Details:
    --------------
    Error using jointBuilder/buildJoint (line 99)
    jointBuilder.buildJoint error: Linear model 'full_dyn' do not exist!
    
    Error in jointBuilderTest>testInputParameters (line 195)
        jb.buildJoint(allParams{1}, 'full_dyn', [],'electric_dyn');
================================================================================
.
================================================================================
Error occurred in jointBuilderTest/testElectricalSubsystem and it did not run to completion.

    ---------
    Error ID:
    ---------
    ''

    --------------
    Error Details:
    --------------
    Error using jointBuilder/buildJoint (line 99)
    jointBuilder.buildJoint error: Linear model 'full_dyn' do not exist!
    
    Error in jointBuilderTest>testElectricalSubsystem (line 241)
        jb.buildJoint(allParams{1}, 'full_dyn', [],'electric_dyn','elTestJoint');
================================================================================
.
Done jointBuilderTest
__________

Failure Summary:

     Name                                         Failed  Incomplete  Reason(s)
    ============================================================================
     jointBuilderTest/testNoFrictionModels          X         X       Errored.
    ----------------------------------------------------------------------------
     jointBuilderTest/testCoulombModels             X         X       Errored.
    ----------------------------------------------------------------------------
     jointBuilderTest/testMultipleNonlinearTerms    X         X       Errored.
    ----------------------------------------------------------------------------
     jointBuilderTest/testInputParameters           X         X       Errored.
    ----------------------------------------------------------------------------
     jointBuilderTest/testElectricalSubsystem       X         X       Errored.

joernmalzahn, 2017-03-07:
Also genericJointTest fails, but that might have the same reason.

closed 2017-03-07 by joernmalzahn

geezIIT, 2017-03-07:
I made a different fix at the same time which is more dynamic (using paths as opposed to string manipulation), so I merged using that in 17cb3ec681242944af45dd1ec19eb2ea88d4b1e6.