geez0x1/CompliantJointToolbox #32: Documentation issue in linear models

geezIIT, 2017-05-03:
The linear models list the input as [tau_m, tau_e], while in fact they are [i, tau_e], i.e. current inputs.

This may also exist in other places.

joernmalzahn, 2017-06-28:
Ok, let's discuss what we wish to have there. Having a consistent torque interface to the mechanical subsystem models would make sense in my opinion. 

This way the electrical subsystems that are actually the torque generators could have voltage as input and torque as output (before or after gearbox, i.e. where should the transmission ratio enter the game?).

joernmalzahn, 2017-09-20:
[image] [Documentation issue in linear models](https://trello.com/c/sww4BqFb/3-documentation-issue-in-linear-models)

geez0x1, 2017-12-11:
I'm currently working on this in the dev_mechanical_torque_inputs branch.

Things that need updating:
- Linear (mechanical) models -> Done
- Electrical models
- Joint model block
- Controllers
- Observers
- A number of tools?

geez0x1, 2017-12-11:
@joernmalzahn Check out the branch if you have time. I think it is a nice improvement. If you have thoughts let me know otherwise I'll merge it into master sometime soon.

geez0x1, 2018-01-02:
@joernmalzahn Have you looked at this? I'm ready to merge it into master.

joernmalzahn, 2018-01-03:
Sorry, didn't manage before the holidays. 
So far it looks good to me. 

Only jointBuilderTest/testElectricalSubsystem fails. Can you look at it?

It would be good to have also a couple of tests for the new input.

joernmalzahn, 2018-01-03:
The error ocurrs in line 277.
         verifyTrue failed.
         --> The value must evaluate to "true".

geez0x1, 2018-01-04:
Hmm, odd. I did go over the tests earlier and updated them. Perhaps I missed this one somehow. Anyway, it's fixed.

I agree more tests are needed. I don't have the time for them now though, perhaps next month. Merging this into master now.

closed 2018-01-04 by geez0x1