geez0x1/CompliantJointToolbox #36: MATLAB library backwards compatiblity

geezIIT, 2018-03-28:
**This is a copy of the original issue in another repo. This is a more appropriate place.**

MATLAB will not (at all, or with complaints after changing some settings) open Simulink library files created in newer versions of MATLAB/Simulink. Case in point (using R2017a):

[image]

Similarly, I cannot open any of the `cjt_library_*.mdl` files in the `dev_doc` branch.

geezIIT, 2018-03-28:
I had a look around the internet (lots of questions about this). I was hoping to find some model property which we could set to force the file sticking to a certain version (e.g. R2016a), but that doesn't appear to exist.

Unchecking the following checkbox does allow me to load the file, although correct functioning is not guaranteed:
[image]
with the following warnings:
[image]

So I'd like to stress that we should save in a compatible way, to avoid such errors. I propose we write a script that checks and if necessary resaves library files in a compatible version, I'm fairly sure that's doable from a script.

geezIIT, 2018-03-28:
Resources:
https://it.mathworks.com/help/simulink/slref/save_system.html
https://stackoverflow.com/questions/42033503/how-to-standardize-simulink-mdl-file-saving-method-in-a-source-controlled-projec (useful while we're at it)
https://it.mathworks.com/help/simulink/slref/simulink.exporttoversion.html#bu5axol-4
https://it.mathworks.com/help/simulink/ug/saving-a-model.html

geezIIT, 2018-03-28:
In the latest commit in `dev_doc` I have added a file `saveCompatible.m`; it contains a list of essential library files (currently incomplete), checks their `SimulinkVersion`, and if that isn't 8.7 (R2016a), re-saves them. It works quite well.

Perhaps we can expand this to a `prepareRelease.m`, which is ran before merging and tagging new releases on master.

Thoughts?

joernmalzahn, 2018-03-28:
Nice work Wes. 

I think the automation is the only way to practically deal with the backwards compatibility mess. During block development it seems rather impossible to remain backwards compatible (involves too many clicks...). However, as a final stage before merging a dev branch into the master it could be conceivable . 

A release preparation toolkit makes sense, also in general.

geez0x1, 2018-07-22:
I've just updated `saveCompatible.m` to include the new examples. Also, I've updated a few other files in the Simulink examples (openLoop.mdl, openLoop_inputFromFile.mdl and controlBlockDevelopment_generic.mdl) to correctly reference the new CJT library files.

@joernmalzahn Perhaps we can include this in the paper version of the paper? Being at least a bit backwards compatible is important imo.

joernmalzahn, 2018-07-24:
You mean to mention our effort for backwards compatibility in the paper?

geez0x1, 2018-07-24:
No I mean to ensure that the version committed for the paper (both on Github and Code Ocean) are backwards compatible and identical.

joernmalzahn, 2018-07-24:
Ok, that's no big deal.

geez0x1, 2018-07-27:
This works and is used.

closed 2018-07-27 by geez0x1