geez0x1/CompliantJointToolbox #37: Sampling times and electrical subsystem

geezIIT, 2018-03-28:
In 96f0024 I have added `Ts_elec`, for the sampling time of the electrical subsystem. The electrical models should use this as their sampling time for computation if input/output delays and possible discretisation.

At the current moment, we have `Ts = 1e-3` and `Ts_elec = 5e-5`. While I think the latter is fine, AFAIK the former is 5e-4 in firmware. Using Simulink RT, it effectively becomes 1e-3 with some delays.

The question is, which one should be default? @joernmalzahn what's your opinion?

joernmalzahn, 2018-03-28:
In my opinion, the small sampling time for the electric subsystem makes only sense, if the electrical dynamics is actually used in the model. 

The story behind the electrical subsystem dynamics is, that as a toolbox user, I wish to investigate the influence of the non-ideal closed current loop dynamics on the torque control performance. To do that, the sampling times must be about two orders of magnitude smaller. 
Each second simulated time will take more more simulation steps and simulation gets slower. Consequently:
a) the Simulink model needs to be configured to use the shorter sampling time
b) I am probably not interested to simulate 10 seconds or more. Everything I want to look at, probably happens within one second.

The story for most users is probably, to check the torque control performance under some conditions / loads, assuming ideal current control. Here the model should use normal sampling time (Ts). Otherwise simulation will take ages.

I agree it might lead to unexpected behavior with respect to the delay blocks, if we switch samping times of the model. The blocks specify a number of samples and not a delay time.

An option could be to have the electrical dynamics subsystem always use Ts_elec, while the mechanical subsystem is always based on Ts? We will need rate transitions at the outputs of the electrical subsystem or the input to the mechanical subsystem.

Another option is to configure the subsystem masks such that they work with a delay time and internally compute the right amount of samples. 

Not fully sure which option is better. The second one could be cleaner.

geez0x1, 2018-03-28:
The option of having the electrical dynamics use `Ts_elec` and the mechanical dynamics use `Ts` is what is currently implemented. Besides computing input/output delay/sampling properly, it also ensures noise values are computed for each new sample.

Alongside, the torque controllers, being discrete time, all use `Ts`, and the current controllers should use `Ts_elec` accordingly. I think that makes sense.

Solver sampling time is a whole other matter indeed; when you start closing current control loops you'll find you need something at around `Ts_elec/10` for it to work properly. That is up to the user.

One question that remains: At the current moment, we have `Ts = 1e-3` in the code. AFAIK this value is 5e-4 in firmware. Using Simulink RT, it effectively becomes 1e-3 with some additional delays. Which should be default? I tend towards 5e-4.

geez0x1, 2018-04-10:
Current controller is fixed, using ZOH now.

joernmalzahn, 2018-07-09:
Is it possible that this did not find its way into the dev_doc branch, where we are currently working (see #51 )

geez0x1, 2018-07-23:
That was a different issue, specific to that model: #51.

closed 2018-07-23 by geez0x1