geez0x1/CompliantJointToolbox #39: d-q current model integration into genericJoint

geezIIT, 2018-03-28:
- `getElectricalDynamicsMatrices` effectively is moot when using this, the d-q current model is nonlinear
- The function could be modified to return an error stating that the joint was built with nonlinear (d-q) current model
- Replacing it with another function is an option, but ill-advised in my opinion as its output requires giving a state and input. Since the joint object itself is currently stateless, I want to avoid this.

geezIIT, 2018-03-28:
Starting 453ab2d there is a fully functioning d-q current model in the library. It is a separate block from the regular electrical dynamics block, but in my opinion that's preferred as the differences between the two are rather large and making something that morphs between them will be very involved, with the result being flexible but very complex.

There is still a minor hack within the block with regards to passing the joint's parameters, but it is more minor than before and none of it is user-facing. This could possibly be improved in the future.

TODO:
- [ ] Update the `jointBuilder` to not provide (or output an error) the `getElectricalDynamicsMatrices` function for joints built with d-q current model.

geez0x1, 2018-03-29:
As far as I can tell this is working properly.

closed 2018-03-29 by geez0x1