geez0x1/CompliantJointToolbox #44: cjt_* param files have very (too) low viscous damping

geez0x1, 2018-04-10:
The `cjt_*` parameter files have 10x smaller viscous damping than the `CE_Med_ds` param file, which is based on datasheet values, and is *still* underdamped compared to experimental data. This makes joint models based on the `cjt_*` parameter files very hard to control.

These parameter values need to be updated according to experimental data; a good start is using accurate datasheet values.

geez0x1, 2018-04-11:
Interestingly, this is mostly masked by the internal gearbox damping when using non-rigid gearbox models. Only when using rigid gearbox models the difference becomes apparent (poles having damping of 4.56e-03 as opposed to 3.12e-02).

geez0x1, 2018-06-21:
In the mean time, the (non-internal) damping parameters for the orange/medium 80:1 actuator have been experimentally identified and those parameters are in the `CE_Med`params in the `CE_Med` branch.

Comparing the relevant sections from `CE_Med` vs `cjt_Orange_80_6000`, also including some other things like inertia, which appear to be pretty close:

```
CE_Med
          I_m: 0.2164
          I_g: 0.0788
          I_l: 1
          k_g: 25000
          k_b: 5714
          d_m: 0.2218
          d_g: 1.9966
          d_l: 0
        d_m_n: 0.2237
        d_g_n: 2.0129
        d_l_n: 0
         d_mg: 300
         d_gl: 0
         d_cm: 0.3538
         d_cg: 3.1838
         d_cl: 0
       d_cm_n: 0.3455
       d_cg_n: 3.1092
       d_cl_n: 0
```

```
cjt_Orange_80_6000
          I_m: 0.2300
          I_g: 0.0717
          I_l: 1.1370e-04
          k_g: 16000
          k_b: 6000
          d_m: 0.0036
          d_g: 0.3840
          d_l: 0
        d_m_n: 0.0036
        d_g_n: 0.3840
        d_l_n: 0
         d_mg: 100
         d_gl: 0
         d_cm: 2.6400
         d_cg: 3.2800
         d_cl: 0
       d_cm_n: 2.6400
       d_cg_n: 3.2800
       d_cl_n: 0
```

joernmalzahn, 2018-06-21:
Hmm not quite sure, if I get this correctly. What values do you wish to see in the toolbox parameter files? What is now the conclusion about friction parameters between CE_Med and cjt_Orange? Should they be copied?

The gearbox stiffness seems actually quite reasonable. Remember that the gearboxes stiffen with increasing torque. Typically the approximation is a three segment piecwise linear one. The toolbox just considers one linear stiffness. 

What are "accurate datasheet parameters". And where would one find them? There are datasheets for parts. Most often motor and gearbox vendors do not provide any value. Sometimes they provide values that are "fictive something" (see Maxon datasheets). Gearboxes have a no load starting torque, which differs depending on which side you apply it.

It might in general be a very tedius thing to search for friction values for all parameter files that are accurate for all real hardware that we use. While this might be sure paradise for us that we are working with the fruit actuators, users of the toolbox that might never actually see a fruit actuator will be fine with ball-park numbers and probably they will play with them anyways. So I do not see an urgency for the next release.

Do you already have any experience so far, how consistent the parameters are across different specimen of the same actuator type?

geez0x1, 2018-06-25:
The wording 'accurate' datasheet values was a bit unfortunate - what I meant was to base unknown parameters on their datasheet values where available. For example, as I did in `CE_Med_ds` (the datasheet version of `CE_Med`:
```
% Friction calculations
% Motor:    Viscous friction estimated directly from datasheet:
%           d_rads = 5.36e-3 * (60/(1000*2*pi))
%           Coulomb friction: d_cm = 0.033 * n
% Gearbox:  Viscous friction estimated from 'no load running torque' at 3500rpm
%           input speed, 20 degrees Celcius, after removing compensation values
%           (table 29.1) and no load starting torque (table 29.2).
%           tau = (16-0.3-2.9)/100; omega = (3500/60)*(2*pi); d = n^2 * tau / omega;
%           'No load starting torque' used for Coulomb friction
%           values: d_cg = 2.9/100 * n
```
This yielded values which later turned out to be about 15% off from experimental values (40% for Coulomb).

I would suggest that `cjt_Orange_` for 80:1 should get its viscous and Coulomb friction parameters (the ones w.r.t. housing) copied from `CE_Med` (=80:1), as `CE_Med` contains experimentally determined values. I would be fine with ballpark values, but the current values are *not* representative, hence I think they should be changed.

Other `cjt_` models could get suitably scaled copies; scaling by `100^2 / 80^2` for the 100:1 Orange and by ratio of datasheet values for the different types (for the time we do not have identified values for those models).

This will at least allow us to get inertia and damping (w.r.t ground) _good enough_ in the models the toolbox generates. Stribeck and internal gearbox dynamics are another beast entirely which I do not intend to capture for the time being.

Inter-sample variance is something I'd like to test but at the moment I not have multiple samples at my disposal (and the ones on the leg have customised mounting adapters causing them not to fit on the bench).

joernmalzahn, 2018-06-26:
>   whether you agree with updating cjt_Orange_ 80:1 using the experimental values;

Fine with me.

>   where the current viscous/Coulomb friction parameters in cjt_* param files came from;

As far as possible, scaled datasheet values. 

>   your thoughts on using scaled copies for the other cjt_* param files in dev_doc.

Agree.

joernmalzahn, 2018-07-09:
Updated parameters in: 165954299a09b3a0ac3fa8182a169902704244b7

Double checked motor friction with datasheets. They are correct.
The Coulomb parameters for the gearbox correspond with the no-load starting torque. 

For the viscous damping, I used the identified values.

joernmalzahn, 2018-07-24:
@geez0x1 Would you be happy to close this now?

geez0x1, 2018-07-24:
I had a look at `cjt_Orange_80`. The parameters are mostly fine, just Coulomb friction is quite high (5.92 Nm vs 3.54 Nm on `CE_Med`), but not crazy unrepresentative. Closing then.

closed 2018-07-24 by geez0x1