geez0x1/CompliantJointToolbox #45: Joint model outputs when using output delays

geez0x1, 2018-04-12:
At the moment when output delays are set >0, *all* fields are delayed. However, for seeing how well something is tracked you might want to have the non-delayed field (e.g. the *noNoise* fields).

geez0x1, 2018-04-13:
On this note, I have updated all controllers to use the `TorqueFiltered` field as opposed to the `Torque` field, which architecturally makes more sense.

However the point w.r.t. output delays still stands. One can manually shift the signal in post-processing, but that is laborious.

joernmalzahn, 2018-04-17:
Hmm, does it make more sense? If I just want to see how the controllers behave in the ideal case, what would I do now?
I am not perfectly happy with the different clones of the same signal in the joint bus, actually.

geez0x1, 2018-04-18:
For example, PD controllers would not have access to a clean torque signal (one can only set derivative filtering in the controller itself). Assuming there's noise available there is no situation you would give the controller an unfiltered signal. If you want to test the ideal case, currently you'd set delays to zero, noise off, and filter frequency to a "very large" value. This last bit I think we can improve, by having a checkbox which effectively removes the filter from the path entirely.

With regards to having different clones, I agree to some extent. There are cases where one would want multiple - e.g. the 'polluted' (noisy, delayed, filtered, etc) signal needs to be passed to the controller, whereas you'd want to have the perfect torque signal to accurately judge obtained performance.

geez0x1, 2018-06-21:
Towards this end I have updated the mechanical joint model block to have all __NoNoise__ output variables denote physical quantities, i.e. unaffected by noise, delays, or filtering. I have also improved the block's help to properly explain all inputs and outputs:

> This block implements the mechanical dynamics of the actuator. Depending on the configuration in which the joint class was built, its dynamics are given by one of the variants provided by the toolbox (see model/linear/). If provided, the linear dynamics are enhanced with nonlinear components.

<b>Inputs</b>
<b>motorTorque</b>: Electromechanical motor torque provided on the rotor. An input delay can be added to this input (e.g. due to communication delays or a very simplified representation of electrical dynamics).
<b>tau_e or q_l_dot</b>: External interaction torque. Depending on the selected linear dynamics model of the joint, this is either the external torque acting on the load or the imposed load velocity (for fixed-output models).

<b>Outputs</b>
<b>jointBus</b>: The various variables and their measurements, with optional delays/noise/quantisation etc. For most outputs, several instances exist. Output variables ending in "NoNoise" denote physical quantities, without noise, filtering, or delays. All other variables are affected by noise and delays (if enabled). Lastly, variables ending in "Filtered" are also low-pass filtered, with cut-off frequency set in the mask.

commit c58026cd (2018-06-21): Updated lib/cjt_library_models.mdl to fix #45
closed 2018-06-21 by geez0x1