geez0x1/CompliantJointToolbox #47: True torque output when the sensor has internal damping

geez0x1, 2018-05-01:
Following some testing with internal component damping where I set `d_gl > 0` I found that the current jointBus does not allow to get the true torque in these cases. In the block the torque fields are computed using only the stiffness contribution `k_b * (q_g - q_l)`.

Combined with #45 , we should probably consider redefining the jointBus.

geez0x1, 2018-06-21:
Some changes were made to the jointBus (in the "NoNoise" variables) following #45.

That fix does not solve this issue however: I believe when we redefine the jointBus in a future version we should have some clear distinction between "sensed" and "physical" variables. Sensed variables obviously are the ones controllers have at their disposal, but for analysis we obviously want to be able to have the physical quantities.

In the context of this issue this means that the "sensed" torque would include only the spring-part of the torque (as well as optional noise, delays, filtering), and the physical quantity would include both stiffness and damping terms of the element.

joernmalzahn, 2018-06-21:
Sounds reasonable to me.

geez0x1, 2018-06-25:
OK. Added enhancement label to this to be included in future refactoring.

geez0x1, 2018-06-25:
To partially fix this issue before future refactoring, in 1046504:
> Added the damping component of the flexible element to the "TorqueNoNoise" output of the jointBus, as all "NoNoise" output variables are said to be physical quantities. Also updated Help of the Simulink library block accordingly. Partially fixes #47.

joernmalzahn, 2018-06-26:
Cool!

geez0x1, 2018-07-22:
This should be open for the refactoring part!

geez0x1, 2019-01-17:
I've fixed a few issues with regards to this in ef5ab51da217727e6611bbbf8bbc86b2a138432b. The `output_fixed` model had an issue in how the damping term was computed in state-space, and in construction of the `jointBus` the damping component was incorrectly included in TorqueFiltered (which is a sensed signal of the torque, and as we consider deflection/stiffness measurements for torque it should not know about the damping component of the flexure).

commit 1046504d (2018-06-25): Added the damping component of the flexible element to the "TorqueNoNoise" output of the jointBus, as all "NoNoise" output variables are said to be physical quantities. Also updated Help of the Simulink library block accordingly. Partially fixes #47.