geez0x1/CompliantJointToolbox #50: Failing unit tests in dev_doc

geez0x1, 2018-06-25:
``
#####################################
TEST SUMMARY
#####################################
----------------
JointBuilder
----------------

ans =

  5×6 table

                          Name                          Passed    Failed    Incomplete    Duration      Details   
    ________________________________________________    ______    ______    __________    ________    ____________

    'cjtJointBuilderTest/testNoFrictionModels'          true      false     false          0.35604    [1×1 struct]
    'cjtJointBuilderTest/testCoulombModels'             true      false     false           0.4606    [1×1 struct]
    'cjtJointBuilderTest/testMultipleNonlinearTerms'    true      false     false          0.28639    [1×1 struct]
    'cjtJointBuilderTest/testInputParameters'           true      false     false         0.092067    [1×1 struct]
    'cjtJointBuilderTest/testElectricalSubsystem'       true      false     false          0.26112    [1×1 struct]

----------------
GenericJoint
----------------

ans =

  4×6 table

                           Name                           Passed    Failed    Incomplete    Duration      Details   
    __________________________________________________    ______    ______    __________    ________    ____________

    'cjtGenericJointTest/testBuiltJoint'                  true      false     false         0.13926     [1×1 struct]
    'cjtGenericJointTest/testConversionSymNum'            true      false     false           5.934     [1×1 struct]
    'cjtGenericJointTest/testSymbolicStateSpace'          true      false     false          3.5677     [1×1 struct]
    'cjtGenericJointTest/testSymbolicTransferFunction'    true      false     false          10.294     [1×1 struct]

----------------
DataSheetGenerator
----------------

ans =

  1×6 table

                          Name                          Passed    Failed    Incomplete    Duration      Details   
    ________________________________________________    ______    ______    __________    ________    ____________

    'cjtDataSheetGeneratorTest/testTorqueSpeedCurve'    false     true      true          79.29       [1×1 struct]

----------------
bodeTest
----------------

ans =

  2×6 table

                 Name                 Passed    Failed    Incomplete    Duration      Details   
    ______________________________    ______    ______    __________    ________    ____________

    'cjtBodeTest/testBasicPlot'       true      false     false         1.5247      [1×1 struct]
    'cjtBodeTest/testResamplePlot'    true      false     false         1.4881      [1×1 struct]

----------------
bode2Test
----------------

ans =

  1×6 table

                Name                 Passed    Failed    Incomplete    Duration      Details   
    _____________________________    ______    ______    __________    ________    ____________

    'cjtBode2Test/testBasicPlots'    true      false     false         1.5139      [1×1 struct]

----------------
SimulinkLibrary
----------------

ans =

  9×6 table

                     Name                      Passed    Failed    Incomplete    Duration      Details   
    _______________________________________    ______    ______    __________    ________    ____________

    'cjtSimulinkLibraryTest/testBuiltJoint'    false     true      true          11.287      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_02'    false     true      true          5.8004      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_03'    false     true      true           1.745      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_04'    false     true      true          3.0596      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_05'    false     true      true          2.4879      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_06'    false     true      true           4.567      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_07'    false     true      true          2.3533      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_08'    false     true      true          4.0973      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_09'    false     true      true          3.0231      [1×1 struct]

----------------
ChecksumTest
----------------

ans =

  4×6 table

                    Name                     Passed    Failed    Incomplete    Duration      Details   
    _____________________________________    ______    ______    __________    ________    ____________

    'cjtChecksumTest/testLoremIpsumArray'    false     true      false           1.1672    [1×1 struct]
    'cjtChecksumTest/testLoremIpsumFile'     false     true      false          0.26024    [1×1 struct]
    'cjtChecksumTest/testMagic10Array'       false     true      false           0.3221    [1×1 struct]
    'cjtChecksumTest/testMagic10File'        true      false     false         0.023008    [1×1 struct]

Passed:     13
Failed:     13
Incomplete: 10
```

joernmalzahn, 2018-07-04:
```
----------------
ChecksumTest
----------------

ans =

  4×6 table

                    Name                     Passed    Failed    Incomplete    Duration      Details   
    _____________________________________    ______    ______    __________    ________    ____________

    'cjtChecksumTest/testLoremIpsumArray'    false     true      false           1.1672    [1×1 struct]
    'cjtChecksumTest/testLoremIpsumFile'     false     true      false          0.26024    [1×1 struct]
    'cjtChecksumTest/testMagic10Array'       false     true      false           0.3221    [1×1 struct]
    'cjtChecksumTest/testMagic10File'        true      false     false         0.023008    [1×1 struct]

Passed:     13
Failed:     13
Incomplete: 10
'''
This one might be the most consequential fail. If the checksum is not computed correctly, the remaining fails might not be meaningful. 
Unfortunateley, I cannot reproduce these on my machine. They all pass. Interestingly all but one seem to fail on yours. It might be a version issue and related to the way the checksums are computed. Let's look into that together.

'''
----------------
SimulinkLibrary
----------------

ans =

  9×6 table

                     Name                      Passed    Failed    Incomplete    Duration      Details   
    _______________________________________    ______    ______    __________    ________    ____________

    'cjtSimulinkLibraryTest/testBuiltJoint'    false     true      true          11.287      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_02'    false     true      true          5.8004      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_03'    false     true      true           1.745      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_04'    false     true      true          3.0596      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_05'    false     true      true          2.4879      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_06'    false     true      true           4.567      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_07'    false     true      true          2.3533      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_08'    false     true      true          4.0973      [1×1 struct]
    'cjtSimulinkLibraryTest/testExample_09'    false     true      true          3.0231  
'''
They fail on my machine as well. These fail, because we have not yet updated the checksums after updating model files, right? 

''''
----------------
DataSheetGenerator
----------------

ans =

  1×6 table

                          Name                          Passed    Failed    Incomplete    Duration      Details   
    ________________________________________________    ______    ______    __________    ________    ____________

    'cjtDataSheetGeneratorTest/testTorqueSpeedCurve'    false     true      true          79.29       [1×1 struct]

''''
It passes on my machine. Two theories about that one:
a) The problem of checksums not being computed properly.
b) Inkscape and / or tex compiler not properly set-up. I should write a script / guide that helps with that.

Can you please provide more infos by running this one:
runtests('cjtDataSheetGeneratorTest/testTorqueSpeedCurve')

geez0x1, 2018-07-09:
Another point is the following:
```
Warning: The character encoding windows-1252 used by model 'simulinkBlockLibraryTest' differs from the MATLAB session
character encoding UTF-8. To check for problems which might corrupt the model use the "Simulink Model File Integrity"
checks in the Model Advisor. 
```
This *should not* have an effect, but with MATLAB you never know.. I'll have a look at the checksumTest ones though, if that doesn't work everything else will fail as well like you mentioned.

geez0x1, 2018-07-10:
checksumTest now fixed (see referenced issue)

geez0x1, 2018-07-10:
Full output from **runtests('cjtDataSheetGeneratorTest/testTorqueSpeedCurve')**:

```
Joint built and written to /data/nextCloud/IIT/SEA_control/CompliantJointToolbox/build/cjt_Orange_80_6000_full_dyn.m
Warning: MATLAB has disabled some advanced graphics rendering features by switching to software OpenGL. For more
information, click here. 

h = 

  Surface (t_{max} [s] to max. temperature) with properties:

       EdgeColor: [0 0 0]
       LineStyle: '-'
       FaceColor: 'flat'
    FaceLighting: 'flat'
       FaceAlpha: 1
           XData: [100×101 double]
           YData: [100×101 double]
           ZData: [100×101 double]
           CData: [100×101 double]

  Show all properties

Datasheet compilation failed. See compiler output for details: 
pdflatex: /home/wesley/MATLAB/R2017a/sys/os/glnxa64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by pdflatex)
pdflatex: /home/wesley/MATLAB/R2017a/bin/glnxa64/libtiff.so.5: no version information available (required by /usr/lib/x86_64-linux-gnu/libpoppler.so.73)

Datasheet compilation failed. See compiler output for details: 
pdflatex: /home/wesley/MATLAB/R2017a/sys/os/glnxa64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by pdflatex)
pdflatex: /home/wesley/MATLAB/R2017a/bin/glnxa64/libtiff.so.5: no version information available (required by /usr/lib/x86_64-linux-gnu/libpoppler.so.73)

Datasheet compilation failed. See compiler output for details: 
pdflatex: /home/wesley/MATLAB/R2017a/sys/os/glnxa64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by pdflatex)
pdflatex: /home/wesley/MATLAB/R2017a/bin/glnxa64/libtiff.so.5: no version information available (required by /usr/lib/x86_64-linux-gnu/libpoppler.so.73)

================================================================================
Error occurred in cjtDataSheetGeneratorTest/testTorqueSpeedCurve and it did not run to completion.

    ---------
    Error ID:
    ---------
    'MATLAB:COPYFILE:OSError'

    --------------
    Error Details:
    --------------
    Error using copyfile
    cp: cannot stat '/data/nextCloud/IIT/SEA_control/CompliantJointToolbox/cjtdsheet.pdf': No such file or directory
    
    Error in dataSheetGenerator/generateDataSheet (line 1189)
                copyfile([fName,'.pdf'],destFName);
    
    Error in cjtDataSheetGeneratorTest>testTorqueSpeedCurve (line 140)
    dsg.generateDataSheet;
================================================================================
.
Done cjtDataSheetGeneratorTest
__________

Failure Summary:

     Name                                            Failed  Incomplete  Reason(s)
    ===============================================================================
     cjtDataSheetGeneratorTest/testTorqueSpeedCurve    X         X       Errored.
    

ans = 

  TestResult with properties:

          Name: 'cjtDataSheetGeneratorTest/testTorqueSpeedCurve'
        Passed: 0
        Failed: 1
    Incomplete: 1
      Duration: 72.5485
       Details: [1×1 struct]

Totals:
   0 Passed, 1 Failed, 1 Incomplete.
   72.5485 seconds testing time.
```

**The problem originates here:**
```
pdflatex: /home/wesley/MATLAB/R2017a/sys/os/glnxa64/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by pdflatex)
pdflatex: /home/wesley/MATLAB/R2017a/bin/glnxa64/libtiff.so.5: no version information available (required by /usr/lib/x86_64-linux-gnu/libpoppler.so.73)
```
It looks like MATLAB is trying to use some outdated stdc++ library that it includes itself, which clashes with my system. I'm not sure *why* it's trying to do that though, since obviously stdc++ is present on the system.

geez0x1, 2018-07-10:
Apparently this issue has been around for 6+ years. Latest thread I could find on it confirms my suspicion:

https://nl.mathworks.com/matlabcentral/answers/329796-issue-with-libstdc-so-6

I'm not sure how to fix it for our users; perhaps a warning will suffice. Messing around with their MATLAB installation is not a good idea :P

geez0x1, 2018-07-10:
After fixing the issue, there are still LaTeX compilation issues on my system. I would guess due to missing LaTeX packages? 

```
Datasheet compilation failed. See compiler output for details: 
This is pdfTeX, Version 3.14159265-2.6-1.40.18 (TeX Live 2017/Debian) (preloaded format=pdflatex)
 restricted \write18 enabled.
entering extended mode
(./cjtdsheet.tex
LaTeX2e <2017-04-15>
Babel <3.18> and hyphenation patterns for 9 language(s) loaded.
(./cjtdsheet.cls
Document Class: cjtdsheet 2016/10/02 version 0.1 CompliantJointToolbox Datashee
t
(./cjtdsheet.cfg) (/usr/share/texlive/texmf-dist/tex/latex/base/article.cls
Document Class: article 2014/09/29 v1.4h Standard LaTeX document class
(/usr/share/texlive/texmf-dist/tex/latex/base/size10.clo))
(/usr/share/texlive/texmf-dist/tex/latex/geometry/geometry.sty
(/usr/share/texlive/texmf-dist/tex/latex/graphics/keyval.sty)
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/ifpdf.sty)
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/ifvtex.sty)
(/usr/share/texlive/texmf-dist/tex/generic/ifxetex/ifxetex.sty))
(/usr/share/texlive/texmf-dist/tex/latex/tools/multicol.sty)
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/ifluatex.sty)
(/usr/share/texlive/texmf-dist/tex/latex/base/fontenc.sty
(/usr/share/texlive/texmf-dist/tex/latex/base/t1enc.def))
(/usr/share/texlive/texmf-dist/tex/latex/fancyhdr/fancyhdr.sty)
(/usr/share/texlive/texmf-dist/tex/latex/graphics/graphicx.sty
(/usr/share/texlive/texmf-dist/tex/latex/graphics/graphics.sty
(/usr/share/texlive/texmf-dist/tex/latex/graphics/trig.sty)
(/usr/share/texlive/texmf-dist/tex/latex/graphics-cfg/graphics.cfg)
(/usr/share/texlive/texmf-dist/tex/latex/graphics-def/pdftex.def)))
(/usr/share/texlive/texmf-dist/tex/latex/amsmath/amsmath.sty
For additional information on amsmath, use the `?' option.
(/usr/share/texlive/texmf-dist/tex/latex/amsmath/amstext.sty
(/usr/share/texlive/texmf-dist/tex/latex/amsmath/amsgen.sty))
(/usr/share/texlive/texmf-dist/tex/latex/amsmath/amsbsy.sty)
(/usr/share/texlive/texmf-dist/tex/latex/amsmath/amsopn.sty))
(/usr/share/texlive/texmf-dist/tex/latex/amsfonts/amssymb.sty
(/usr/share/texlive/texmf-dist/tex/latex/amsfonts/amsfonts.sty))
(/usr/share/texlive/texmf-dist/tex/latex/tools/tabularx.sty
(/usr/share/texlive/texmf-dist/tex/latex/tools/array.sty))
(/usr/share/texlive/texmf-dist/tex/latex/booktabs/booktabs.sty)
(/usr/share/texlive/texmf-dist/tex/latex/graphics/color.sty
(/usr/share/texlive/texmf-dist/tex/latex/graphics-cfg/color.cfg))
(/usr/share/texlive/texmf-dist/tex/latex/colortbl/colortbl.sty)
(/usr/share/texlive/texmf-dist/tex/latex/caption/caption.sty
(/usr/share/texlive/texmf-dist/tex/latex/caption/caption3.sty))
(/usr/share/texlive/texmf-dist/tex/latex/titlesec/titlesec.sty))
No file cjtdsheet.aux.
*geometry* driver: auto-detecting
*geometry* detected driver: pdftex
(/usr/share/texlive/texmf-dist/tex/context/base/mkii/supp-pdf.mkii
[Loading MPS to PDF converter (version 2006.09.02).]
) (/usr/share/texlive/texmf-dist/tex/latex/oberdiek/epstopdf-base.sty
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/infwarerr.sty)
(/usr/share/texlive/texmf-dist/tex/latex/oberdiek/grfext.sty
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/kvdefinekeys.sty
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/ltxcmds.sty)))
(/usr/share/texlive/texmf-dist/tex/latex/oberdiek/kvoptions.sty
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/kvsetkeys.sty
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/etexcmds.sty)))
(/usr/share/texlive/texmf-dist/tex/generic/oberdiek/pdftexcmds.sty)
(/usr/share/texlive/texmf-dist/tex/latex/latexconfig/epstopdf-sys.cfg))
Underfull \hbox (badness 10000) in paragraph at lines 6--7

(/usr/share/texlive/texmf-dist/tex/latex/amsfonts/umsa.fd)
(/usr/share/texlive/texmf-dist/tex/latex/amsfonts/umsb.fd)
Underfull \hbox (badness 10000) in paragraph at lines 50--50
[]|\T1/cmr/m/n/10 Nm

[... 291 lines cut ...]
 )
(see the transcript file for additional information) </home/wesley/.texlive2017
/texmf-var/fonts/pk/ljfour/jknappen/ec/ecti1000.600pk> </home/wesley/.texlive20
17/texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1200.600pk> </home/wesley/.texlive
2017/texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1728.600pk> </home/wesley/.texli
ve2017/texmf-var/fonts/pk/ljfour/jknappen/ec/ecbi1000.600pk> </home/wesley/.tex
live2017/texmf-var/fonts/pk/ljfour/jknappen/ec/ecrm1000.600pk> </home/wesley/.t
exlive2017/texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx1000.600pk> </home/wesley/
.texlive2017/texmf-var/fonts/pk/ljfour/jknappen/ec/ecbx2488.600pk></usr/share/t
exlive/texmf-dist/fonts/type1/public/amsfonts/cm/cmmi10.pfb></usr/share/texlive
/texmf-dist/fonts/type1/public/amsfonts/cm/cmmi7.pfb></usr/share/texlive/texmf-
dist/fonts/type1/public/amsfonts/cm/cmr10.pfb></usr/share/texlive/texmf-dist/fo
nts/type1/public/amsfonts/cm/cmr7.pfb></usr/share/texlive/texmf-dist/fonts/type
1/public/amsfonts/cm/cmsy7.pfb>
Output written on cjtdsheet.pdf (6 pages, 1619957 bytes).
SyncTeX written on cjtdsheet.synctex.
Transcript written on cjtdsheet.log.

-------
Warning: If the above error references libstdc++, this issue arises due to MATLAB including libraries which are too old for your system. Try removing the following symlinks:
- /MATLAB/R2017a/sys/os/glnxa64/libstdc++.so.6
- /MATLAB/R2017a/bin/glnxa64/libtiff.so.5
See also: https://mathworks.com/matlabcentral/answers/329796-issue-with-libstdc-so-6
-------
```

geez0x1, 2018-07-23:
Indeed, the \circ thing (probably a new one added later) was the issue. New output from
`runtests('cjtDataSheetGeneratorTest/testTorqueSpeedCurve')`:

```
Done cjtDataSheetGeneratorTest
__________

ans = 

  TestResult with properties:

          Name: 'cjtDataSheetGeneratorTest/testTorqueSpeedCurve'
        Passed: 1
        Failed: 0
    Incomplete: 0
      Duration: 63.3217
       Details: [1×1 struct]

Totals:
   1 Passed, 0 Failed, 0 Incomplete.
   63.3217 seconds testing time.
```

Fixed.

closed 2018-07-23 by geez0x1