geez0x1/CompliantJointToolbox #52: Warning caused by : Transport Delay block of mechanical subsystem

joernmalzahn, 2018-07-11:
The Simulink files throw this warning on the console:
```
Warning: When delay time is set to zero, the transport delay block
'Ex_08_PD_Feedforward_ClosedLoopDOB/JointMechanicalSubsystem/input_delay' is automatically set to support direct
feedthrough. This may cause an algebraic loop. A Memory Block can be used in place of the Transport Delay to break
the loop 
``` 
We should implement the delay, such that Simulink does not feel obliged to throw this warning.

geez0x1, 2018-07-22:
Does this only happen in the DOBs? Probably a consequence of discretisation methods of the filters then.

geez0x1, 2018-07-23:
Fixed in 2dc8ada.

The delay value inside the actual delay block is computed so that it is >0 when delay is DISABLED,
to avoid MATLAB throwing warnings unnecessarily. Depending on whether delay is actually enabled the signal is routed around the block or through it.

closed 2018-07-23 by geez0x1