geez0x1/CompliantJointToolbox #53: Kalman observer and Luenberger observer do not find stable gains

joernmalzahn, 2018-07-12:
Both, the Kalman block and the Luenbeger block throw this error when inserting the joint class parameter:

[image]

The guess is that the blocks have not been updated after recent interface changes (input / output dimensionality). This way the blocks are unusable.

geez0x1, 2018-07-24:
This is an interesting issue. The systems used in the examples (`output_fixed`) are not observable due to the secondary input.

geez0x1, 2018-07-25:
This has been fixed by somewhat of a workaround for the Kalman observer, and properly for the Luenberger observer. Both examples and unit tests work now: b6819be16f3a01ada97b4e49b54c416c9d4e20d1...c5b0095f29442c74fc8b436ce251762a8bc2bc7b.

closed 2018-07-25 by geez0x1