geez0x1/CompliantJointToolbox #70: New User For CompliantJointToolbox

Mohammed20PHD, 2019-09-20:
Hi, I am New user for CompliantJointToolbox And I tried to run the toolbox with windows 10 matlab2018a, but I got some error as shown below
Can you help me to fix this problem??

>> setCJTPaths
Added Compliant Joint Toolbox (CJT) paths.
---------------------------------------------------------------------------
>> Run
Error using jointBuilder/buildJoint (line 206)
jointBuilder.buildJoint error: Nonlinear model 'coulomb' does not exist!

Error in Take_A_Tour (line 70)
jb.buildJoint ( 'example_params' , ... parameters from example_params.m
---------------------------------------------------------------------
>> cjtExamples
Reference to non-existent field 'activeListbox'.

Error in cjtExamples>localRun_callback (line 267)
listboxTag = get(fig_handles.activeListbox,'Tag');
 
Error while evaluating UIControl Callback.
---------------------------------------------------------------------------
>> Take_A_Tour
Added Compliant Joint Toolbox (CJT) paths.

%% Motor parameters
% Save these lines in example_params.m
% for later use in other examples.
%
% Motor rotor inertia [kg m^2]
params.('I_m') = 9.407000e-02;     
% Motor Damping [Nms/rad]
params.('d_m') = 2.188643e-03;
% Torque constant [Nm/A]
params.('k_t') = 4.100000e-02;

%% Gear parameters
% Gear transmission ratio [.]
params.('n')   = 100;

%% Sensor parameters
% Sensor inertia [kg m^2]
params.('I_l') = 1.137000e-04;
% Sensor stiffness [Nm/rad]
params.('k_b') = 21000;     

jb = 

  jointBuilder with properties:

           buildDir: '..\results'
          overwrite: 0
           basePath: 'C:\Users\mal\MATLAB\CompliantJointToolbox-master\CompliantJointToolbox-master'
    nonlinModelPath: 'C:\Users\mal\MATLAB\CompliantJointToolbox-master\CompliantJointToolbox-master\model\nonlinear'
       linModelPath: 'C:\Users\mal\MATLAB\CompliantJointToolbox-master\CompliantJointToolbox-master\model\linear'

Error using jointBuilder/buildJoint (line 206)
jointBuilder.buildJoint error: Nonlinear model 'coulomb' does not exist!

Error in Take_A_Tour (line 70)
jb.buildJoint ( 'example_params' , ... parameters from example_params.m

geez0x1, 2019-09-23:
Dear Mohammed, thanks for reporting your issue. Let's see if we can find the cause. As far as I can see you are dealing with two issues:
- `jointBuilder.buildJoint error: Nonlinear model 'coulomb' does not exist!` in `Take_A_Tour`, and
- ` Reference to non-existent field 'activeListbox'.` in `cjtExamples`.
Let's focus on the first one first, since it might actually lead to the second one. Can you verify that the following file exists in your CJT installation: `models/nonlinear/coulomb.m`? I assume you are on the master branch?

Mohammed20PHD, 2019-09-23:
Dear Dr. RoozingThank you for your reply, I checked the file in the path  models/nonlinear/coulomb.m and I found the file. In the attached file below
Best Regards

 
 

   

  
%COULOMB Calculate Coulomb friction torques
%
% [ tau, y ] = coulomb(jointObj, x)
%
% jointObj is the instance of the joint class object for which this
% function has been called.
%
%
% Inputs::
%   x: state vector depending on the model type as
%     x = [q_m; q_g; q_l; q_m_dot; q_g_dot, q_l_dot'];  full_dyn
%     x = [q_g, q_l, q_g_dot, q_l_dot]'                 rigid_gearbox
%     x = [q_m, q_g, q_l, q_m_dot, q_g_dot]'            output_fixed
%     x = [q_g, q_l, q_g_dot]'                          output_fixed_rigid_gearbox
%     x = [q_g, q_g_dot]'                               rigid
%
% Outputs::
%   tau: friction torque
%   y:   output components
%
% Notes::
%
%
% Examples::
%
% Author::
%  Joern Malzahn
%  Wesley Roozing
%
% See also full_dyn, coulomb_asym, viscous_asym.

% Copyright (C) 2016, by Joern Malzahn, Wesley Roozing
%
% This file is part of the Compliant Joint Toolbox (CJT).
%
% CJT is free software: you can redistribute it and/or modify
% it under the terms of the GNU General Public License as published by
% the Free Software Foundation, either version 3 of the License, or
% (at your option) any later version.
%
% CJT is distributed in the hope that it will be useful, but WITHOUT ANY
% WARRANTY; without even the implied warranty of MERCHANTABILITY or
% FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public
% License for more details.
%
% You should have received a copy of the GNU General Public License
% along with CJT. If not, see <http://www.gnu.org/licenses/>.
%
% For more information on the toolbox and contact to the authors visit
% <https://github.com/geez0x1/CompliantJointToolbox>

function [ tau, y ] = coulomb(obj, x)
    
    % Preallocate coefficient vector
    c = zeros(size(x));
    
    % Build coefficient vector
    if (strcmp(obj.modelName, 'full_dyn'))
        c = [0, 0, 0, obj.d_cm, obj.d_cg, obj.d_cl]';
        
    elseif (strcmp(obj.modelName, 'rigid_gearbox'))
        c = [0, 0, obj.d_cm + obj.d_cg, obj.d_cl]';
        
    elseif (strcmp(obj.modelName, 'output_fixed'))
        c = [0, 0, 0, obj.d_cm, obj.d_cg]';
        
    elseif (strcmp(obj.modelName, 'output_fixed_rigid_gearbox'))
        c = [0, 0, obj.d_cm + obj.d_cg]';
        
    elseif (strcmp(obj.modelName, 'rigid'))
        c = [0, obj.d_cm + obj.d_cg + obj.d_cl].';
        
    end

    % If we have values over time per variable, pad the coefficients
    if (size(x,2) > 1)
        c = padarray(c, [1 size(x,2)-1], 'replicate', 'post');
    end

    % Calculate Coulomb friction torques
    tau = -c .* tanh(500 * x);
    
    % No nonlinear output components
    y = zeros(7,1);

end

geez0x1, 2019-09-24:
Dear Mohammed, thanks for checking. If the file exists, it appears to not be found due to another reason. Can you verify the following two things:
- Are you trying to execute these steps from the **main directory** of CJT, i.e. where the file `setCJTPaths.m` is located?
- If so, what output does `help coulomb` produce?

Mohammed20PHD, 2019-10-23:
@geez0x1  I want to ask you I work with QBmove advance actuator, 
How can I benefit from this toolbox with the QBmove actuator in designing a controller? 
any advice or recommendations

Thank you very much

joernmalzahn, 2019-10-23:
@Mohammed20PHD: I have never worked with the QBmove actuator. But if you are able to communicate with it through the means Mathworks Simulink RealTime offers, then you will generally be able to deploy all controllers of the compliant joint toolbox with it.

If you do not have the chance to communicate with the actuator through this way, you can still benefit from the toolbox by creating a parameter file to generate simulation models and do simulation studies with various controllers, inputs and load conditions. Then you can prototype new controllers and simulate applications to make educated decisions on what to eventually deploy on the real hardware.

This is essentially how we work. Hope that helps.

joernmalzahn, 2019-10-23:
I'll close this as it seems to be resolved.

closed 2019-10-23 by joernmalzahn

Mohammed20PHD, 2019-10-23:
@joernmalzahn Thank you very much for a nice answer, 
Yes, I can use block Simulink Math to work with QBmove and I will try to deploy all controller of the compliant joint toolbox with it.
Best Regards