geez0x1/CompliantJointToolbox #71: cjtExamples initial state causes issues due to no list item selected

geez0x1, 2019-09-23:
From a user's report in #70:

```
>> cjtExamples
Reference to non-existent field 'activeListbox'.

Error in cjtExamples>localRun_callback (line 267)
listboxTag = get(fig_handles.activeListbox,'Tag');
 
Error while evaluating UIControl Callback.
```

The issue arises in `cjtExamples` on lines 239 in `localOpen_callback` and 267 in `localRun_callback`, callbacks for the "Open Example" and "Run Example" buttons. The `activeListBox` field is not defined if the user has not interacted with the fields yet.

Solution:
The most preferable solution would be to grey out the buttons, and to have nothing selected (visually) upon startup. Then, as soon as the user selects something, the buttons become available.
Perhaps we can ensure that when the user selects something, the other list box completely deselects, for better UX.

@joernmalzahn : FYI. I'll try to fix this soon but since you're better familiar with this code, feel free to jump in :)

Mohammed20PHD, 2019-09-23:
Dear Mohammed, thanks for reporting your issue. Let's see if we can find the cause. As far as I can see you are dealing with two issues:

jointBuilder.buildJoint error: Nonlinear model 'coulomb' does not exist! in Take_A_Tour, and
Reference to non-existent field 'activeListbox'. in cjtExamples.
Let's focus on the first one first, since it might actually lead to the second one. Can you verify that the following file exists in your CJT installation: models/nonlinear/coulomb.m? I assume you are on the master branch?
-----------------------------------------------------
Dear Dr. Roozing
Thank you for your reply, I checked the file in the path  models/nonlinear/coulomb.m and I found the file.
Kindly, Can you please guide me to the next step to run the compliant jointToolbox?

Or Show me by video the steps to run it from the first step if that possible?

thank you very much

geez0x1, 2019-09-24:
Fixed by #72 and #73.

closed 2019-09-24 by geez0x1