geez0x1/CompliantJointToolbox #79: How to figure Torque-frequency diagram for different load inertias?

junjun55, 2022-03-09:
I want to learn about  the feasible torque bandwidth for electrically driven torque controllable . Can the simulation analysis code of this article “What is the Torque Bandwidth of this Actuator?”be added?

joernmalzahn, 2022-03-09:
Hi Junjun, 

not sure, if I get your question right. Please have a look at [this part of the documentation](https://github.com/geez0x1/CompliantJointToolbox/wiki/Torque-Bandwidth-Map). It details the torque bandwidth maps. Probably the second graph is what you would like to achieve?

For the code, have a look at `dataSheetGenerator.m` starting from line 377: 

```
 function h = drawTorqueFrequencyCurveLoad(this, fMax, plotNormalized)
            % DRAWTORQUEFREQUENCYCURVELOAD Theoretically feasible torque bandwidth under varying load inertia.
            %
            % Creates a contour plot that displays the -3dB cut-off frequency for the torque transfer function magnitude 
            % when the actuator generates peak torque. The contour lines are color coded on a logarithmic scale and 
            % correspond to different load inertiae in relation to the combined motor and gear inertia.
            %
            % A very light load will causes the actuator to reach the speed generation limit before a substantial torque
            % level can be established. A heavy load results in torques building up rapidly, with little actuator 
            % motion.
            %
            % The red curve corresponds to an infinite load inertia, that is equivalent to a locked output. The balck
            % solid curve corresponds to the load inertia being equal to the combined motor/gear inertia.
            % 
            % The plot includes the bandwidth limit due to the interplay of spring stiffness and back-EMF generation. 
            % The red area indicates the reachable operating conditions under the assumption of negligible spring 
            % damping. The zone expands with growing sping damping as indicated by black dotted curves. 
            %
            % One axis displays the final temperature at the given
            % operating condition. The second axis displays the time to
            % reach the maximum allowed temperature.
            %
            %   h = drawTorqueFrequencyCurveLoad(this, fMax, plotNormalized)
            %
            % Inputs:
            %   fMax           - The frequency in Hz up to which the magnitude is analyzed (default 100 Hz).
            %   plotNormalized - If true, the frequency axis is normalized to the frequency sqrt( k_ml/(I_m + I_g) ),
            %                    while the torque axis is normalized to the peak torque t_p (default 0). 
            %
            % Outputs:
            %   hContour: Handle to the contour plot
            %   
            %
            % Notes::
            %   For further reading, see: 
            %   J. Malzahn, N. Kashiri, W. Roozing, N. Tsagarakis and D. Caldwell, "What is the torque bandwidth of 
            %   this actuator?," 2017 IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS), 
            %   Vancouver, BC, 2017, pp. 4762-4768. doi: 10.1109/IROS.2017.8206351
```

junjun55, 2022-03-10:
> Hi Junjun,
>
> not sure, if I get your question right. Please have a look at [this part of the documentation](https://github.com/geez0x1/CompliantJointToolbox/wiki/Torque-Bandwidth-Map). It details the [...]

>

junjun55, 2022-03-10:
> > 嗨俊俊，
> > 不确定，如果我答对了你的问题。请查看[文档的这一部分](https://github.com/geez0x1/CompliantJointToolbox/wiki/Torque-Bandwidth-Map)。它详细介绍了扭矩带宽图。也许第二张图是你想要实现的目标？
> > 有关代码，请查看从第 377 行开始：`dataSheetGenerator.m`
> > ```
> >  function h = drawTorqueFrequencyCurveLoad(this, fMax, plotNormalized)
> >             % DRAWTORQUEFREQUENCYCURVELOAD Theoretically feasible torque bandwidth under varying load inertia.
> >             %
> >             % Creates a contour plot that displays the -3dB cut-off frequency for the torque transfer function magnitude 
> >             % when the actuator generates peak torque. The contour lines are color coded on a logarithmic scale and 
> >             % correspond to different load inertiae in relation to the combined motor and gear inertia.
> >             %
> >             % A very light load will causes the actuator to reach the speed generation limit before a substantial torque
> >             % level can be established. A heavy load results in torques building up rapidly, with little actuator 
> >             % motion.
> >             %
> >             % The red curve corresponds to an infinite load inertia, that is equivalent to a locked output. The balck
> >             % solid curve corresponds to the load inertia being equal to the combined motor/gear inertia.
> >             % 
> >             % The plot includes the bandwidth limit due to the interplay of spring stiffness and back-EMF generation. 
> >             % The red area indicates the reachable operating conditions under the assumption of negligible spring 
> >             % damping. The zone expands with growing sping damping as indicated by black dotted curves. 
> >             %
> >             % One axis displays the final temperature at the given
> >             % operating condition. The second axis displays the time to
> >             % reach the maximum allowed temperature.
> >             %
> >             %   h = drawTorqueFrequencyCurveLoad(this, fMax, plotNormalized)
> >             %
> >             % Inputs:
> >             %   fMax           - The frequency in Hz up to which the magnitude is analyzed (default 100 Hz).
> >             %   plotNormalized - If true, the frequency axis is normalized to the frequency sqrt( k_ml/(I_m + I_g) ),
> >             %                    while the torque axis is normalized to the peak torque t_p (default 0). 
> >             %
> >             % Outputs:
> >             %   hContour: Handle to the contour plot
> >             %   
> >             %
> >             % Notes::
> >             %   For further reading, see: 
> >             %   J. Malzahn, N. Kashiri, W. Roozing, N. Tsagarakis and D. Caldwell, "What is the torque bandwidth of 
> >             %   this actuator?," 2017 IEEE/RSJ International Conference on Intelligent Robots and Systems (IROS), 
> >             %   Vancouver, BC, 2017, pp. 4762-4768. doi: 10.1109/IROS.2017.8206351
> > ```
> 
> >

That's it!  Many thanks for your clarification! Your research is very worthy learning for me!